kklem0/Off-Grid-Smart-RV #22: Install solar panels

clementhk, 2020-03-05:


clementhk, 2020-03-05:
We should probably make the panels a bit higher than the windows on the roof so sunlight could still goes through the panels into the RV.
But if we're not gonna open the windows on the rooftop, we could probably change them into glasses, so the sunlight could still come through, then we don't need the panels to be installed too high, though we might still need to do it for the exhaust fan in the toilet, if we don't install it on the wall.

clementhk, 2020-03-05:
Maybe SunPower 390W SPR-MAX3-390 (4,000 NOK) x 4, but wasting a bit of space, re-thinking.

clementhk, 2020-03-10:
Add wind deflector?

clementhk, 2020-03-10:
? https://zerohomebills.com/product/lg-neon2-lg400n2t-j5-400w-bifacial-transparent-solar-panel/

clementhk, 2020-03-10:
? https://zerohomebills.com/product/sunpower-390w-spr-max3-390-mono-solar-panel/

clementhk, 2020-03-10:
? https://www.sonnenshop.de/solarmodule/sunpower-400w-maxeon-3

klem-everywhere, 2021-06-07:
Looking into flexible panels as the RV is a bit overweight now..

klem-everywhere, 2021-06-07:
https://sunpower.maxeon.com/int/sites/default/files/2019-10/sunpower-flexible-panel-170W-6x8.pdf
Potentially can install 10 of these, with a little bit of extra space maybe for StarLink, around 550mm.
But the edges will be stacking, need to look into that.
Here's a guy installing it nicely. https://www.youtube.com/watch?v=uJIEXhnF8MU

klem-everywhere, 2021-06-07:
There's also a smaller version.
https://us.sunpower.com/sites/default/files/sunpower-flexible-panel-170w-4x12_0.pdf